turtlebot/turtlebot_arm #16: Problem with pick and place

xuxinwang, 2015-04-15:


corot, 2015-04-15:
Hi, are you using [my fork](https://github.com/corot/turtlebot_arm)? Because I mostly rewrote the pick and place demo. As far as I can remember, it was working too with the official turtlebot_arm repo, but I suggest you to try my fork instead: I added interesting stuff, as support for PhantomX Pincher arm and gazebo simulation.

If it still fails, the problem must be in moveit. I remember I had similar problems to the one you described and it got fixed with a patch to moveit kinematic solver, and maybe that patch was not backported to hydro. Tell me if it still fails and I'll dig in old commits to find it.

Good luck!

xuxinwang, 2015-04-15:
Thank you for your reply.
I am sure about that I git clone from https://github.com/corot/turtlebot_arm and I test it today.
I change pick 5 times to 50 times but it cannot find the IK solution. 
I will be appreciate if you can find the problem.

corot, 2015-04-15:
mmm... I cannot find any difference between indigo and hydro that can clearly explain the problem. If ever, [this commit](https://github.com/ros-planning/moveit_core/commit/14b020bc616628b50ed383f537af91ee33ac3fd4), but I'm not sure at all. The [problem I was talking about](https://groups.google.com/forum/#!searchin/moveit-users/jorge$20santos$20turtlebot/moveit-users/KNlCGLXb46M/pYWAMsb2KYYJ) was never present on Hydro!
So... first of all, what do you run exactly? I'm testing just now to start turtlebot_arm_moveit_config/demo.launch and turtlebot_arm_moveit_demos/pick_and_place.py and it fails due to a missing parameter (max_opening and min_opening). I just provided default values in the python script and all works ok. Please try to set move_group.launch argument "info" as true. You should see log messages like those:
```
Orientation constraint satisfied for link 'gripper_link'. Quaternion desired: 0.016305 0.047160 -0.326387 0.943918, quaternion actual: 0.015930 0.043683 -0.330987 0.942489, error: x=0.002010, y=0.006832, z=0.009532, tolerance: x=0.010000, y=0.010000, z=0.010000
[ INFO] [1429138635.865571045]: Position constraint satisfied on link 'gripper_link'. Desired: 0.328138, -0.152130, 0.470492, current: 0.327626, -0.152349, 0.470296
[ INFO] [1429138635.865594612]: Differences 0.000512221 0.000219431 0.000195653
[ INFO] [1429138635.868064307]: Orientation constraint satisfied for link 'gripper_link'. Quaternion desired: 0.016305 0.047160 -0.326387 0.943918, quaternion actual: 0.014533 0.052831 -0.329728 0.942484, error: x=0.006648, y=0.009603, z=0.007565, tolerance: x=0.010000, y=0.010000, z=0.010000
[ INFO] [1429138635.868096182]: Position constraint satisfied on link 'gripper_link'. Desired: 0.328138, -0.152130, 0.470492, current: 0.327725, -0.151714, 0.470936
[ INFO] [1429138635.868117098]: Differences 0.000413304 -0.000415407 -0.000444162
[ INFO] [1429138635.868387169]: Orientation constraint satisfied for link 'gripper_link'. Quaternion desired: 0.016305 0.047160 -0.326387 0.943918, quaternion actual: 0.017239 0.050971 -0.330475 0.942280, error: x=0.000250, y=0.007827, z=0.008823, tolerance: x=0.010000, y=0.010000, z=0.010000
```
But saying constraint violated. Try also to increase the goal tolerance:
```
arm.set_goal_position_tolerance(0.04)
arm.set_goal_orientation_tolerance(0.1)
```
though current values are already generous.

double344931987, 2015-04-16:
thanks a lot.
It works

corot, 2015-04-16:
what did you changed?

xuxinwang, 2015-04-16:
I use your new pick_and_place.py 
Thanks a lot

xuxinwang, 2015-04-17:
Sorry to trouble you.
However, I have another question.
In the log messages, it said tolerance: x=0.010000, y=0.010000, z=0.010000
However, you set

arm.set_goal_position_tolerance(0.04)
arm.set_goal_orientation_tolerance(0.1)

Could you please tell me what the difference or how to set the
tolerance in log messages

corot, 2015-04-17:
Not sure.... maybe you can ask the question on moveit google group.

I suppose I can close this issue, right?

closed 2015-04-17 by corot