turtlebot/turtlebot_arm #21: Pincher gripper collision boxes incorrect

Zabot, 2017-08-23:
The collision boxes for the pincher gripper are totally inaccurate, leading to collision boxes clipping into each other and generally misbehaved simulation behavior.

[image]
Here you can see how the collision boxes don't cover the fingers and instead clip into each other and the base. The base's collision box is also offset a bit.

corot, 2017-08-24:
Hi @Zabot, tentative fix on branch issue_21. As I have zero-experience with the pitcher gripper, can you give it a try before merging? I can only tell that looks ok on RViz

Zabot, 2017-08-25:
Yup, looks good to me. Boxes are aligned with geometry and the links aren't clipping anymore.

closed 2017-08-25 by corot