turtlebot/turtlebot_arm #29: The complete state of the robot is not yet known.  Missing gripper_joint

joyyang1215, 2017-09-13:
When I run the **arm.launch** with phantomx pincher arm and run **arbotix_gui**, I find that only the gripper could not be controlled, other joints work fine.

Also, when I run **turtlebot_arm_moveit.launch**, the following error messages show up.
`[ WARN] [1505302333.424168285]: The complete state of the robot is not yet known.  Missing gripper_joint
`

The main problem is that I could not control the gripper.
Any help?

joyyang1215, 2017-09-13:
I've checked that all the servos work, when I use **dynaManager** program.

Zabot, 2017-09-25:
Have you set the environment variable`TURTLEBOT_ARM1` to `pincher`?

joyyang1215, 2017-09-28:
Thanks Zabot,
I've added
`export TURTLEBOT_ARM1=pincher`
in the ~/.bashrc file

joyyang1215, 2017-09-28:
It works when I use [corb555/arbotix_ros](https://github.com/corb555/arbotix_ros).

Zabot, 2017-10-03:
I suspect this is a bug in the arbotix config files. I can observe the full range of motion of the gripper in simulation when I'm not using `arbotix_ros`. Haven't tried to reproduce on hardware yet.

closed 2017-11-09 by joyyang1215

jssf1020, 2018-05-22:
hey bro,maybe you should change your windowx_arm.yaml, the configure of your gripper has something wrong,try to change it.