turtlebot/turtlebot_arm #31: Catkin_Make error with indigo ???

Abduoit, 2017-09-24:
Hi, 

I am using ros indigo, I tried many attempts to solve this issue but I still get same error. I already tried [this link](https://github.com/turtlebot/turtlebot_arm/pull/26/commits/71878822b12c826135cffccbf9b24e33f108e572)
but same error still there. 

```
[ 98%] Building CXX object turtlebot_arm/turtlebot_arm_block_manipulation/CMakeFiles/pick_and_place_action_server.dir/src/pick_and_place_action_server.cpp.o
[100%] [100%] Building CXX object turtlebot_arm/turtlebot_arm_object_manipulation/CMakeFiles/object_interactive_manip_action_server.dir/src/interactive_manipulation_action_server.cpp.o
Building CXX object turtlebot_arm/turtlebot_arm_object_manipulation/CMakeFiles/object_detection_action_server.dir/src/object_detection_action_server.cpp.o
/home/imr/catkin_ws/src/turtlebot_arm/turtlebot_arm_block_manipulation/src/pick_and_place_action_server.cpp:37:62: fatal error: moveit/move_group_interface/move_group_interface.h: No such file or directory
 #include <moveit/move_group_interface/move_group_interface.h>
                                                              ^
compilation terminated.
Linking CXX shared library /home/imr/catkin_ws/devel/lib/libturtlebot_arm_moveit_ikfast_kinematics_plugin.so
[100%] Built target turtlebot_arm_moveit_ikfast_kinematics_plugin
Scanning dependencies of target object_manipulation_test_actions
[100%] Building CXX object turtlebot_arm/turtlebot_arm_object_manipulation/CMakeFiles/object_manipulation_test_actions.dir/test/object_manipulation_actions.cpp.o
In file included from /opt/ros/indigo/include/ros/ros.h:40:0,
                 from /home/imr/catkin_ws/src/turtlebot_arm/turtlebot_arm_object_manipulation/src/interactive_manipulation_action_server.cpp:32:
/home/imr/catkin_ws/src/turtlebot_arm/turtlebot_arm_object_manipulation/src/interactive_manipulation_action_server.cpp: In member function ‘bool turtlebot_arm_object_manipulation::InteractiveManipulationServer::addMarker(const CollisionObject&, bool)’:
/home/imr/catkin_ws/src/turtlebot_arm/turtlebot_arm_object_manipulation/src/interactive_manipulation_action_server.cpp:227:35: error: ‘pose2str3D’ is not a member of ‘mtk’
              marker.name.c_str(), mtk::pose2str3D(marker.pose).c_str(), marker.scale);
                                   ^
In file included from /opt/ros/indigo/include/ros/ros.h:40:0,
                 from /home/imr/catkin_ws/src/turtlebot_arm/turtlebot_arm_object_manipulation/src/object_detection_action_server.cpp:35:
/home/imr/catkin_ws/src/turtlebot_arm/turtlebot_arm_object_manipulation/src/object_detection_action_server.cpp: In member function ‘void turtlebot_arm_object_manipulation::ObjectDetectionServer::goalCB()’:
/home/imr/catkin_ws/src/turtlebot_arm/turtlebot_arm_object_manipulation/src/object_detection_action_server.cpp:218:23: error: ‘pose2str3D’ is not a member of ‘mtk’
                       mtk::pose2str3D(obj.pose.pose.pose).c_str(), bin.id, mtk::pose2str3D(bin.getCentroid()).c_str(),
                       ^
/home/imr/catkin_ws/src/turtlebot_arm/turtlebot_arm_object_manipulation/src/object_detection_action_server.cpp:218:76: error: ‘pose2str3D’ is not a member of ‘mtk’
                       mtk::pose2str3D(obj.pose.pose.pose).c_str(), bin.id, mtk::pose2str3D(bin.getCentroid()).c_str(),
                                                                            ^
/home/imr/catkin_ws/src/turtlebot_arm/turtlebot_arm_object_manipulation/src/object_detection_action_server.cpp:229:65: error: ‘pose2str3D’ is not a member of ‘mtk’
           ROS_DEBUG("Object with pose [%s] added to a new bin", mtk::pose2str3D(obj.pose.pose.pose).c_str());
                                                                 ^
/home/imr/catkin_ws/src/turtlebot_arm/turtlebot_arm_object_manipulation/src/object_detection_action_server.cpp: In member function ‘void turtlebot_arm_object_manipulation::ObjectDetectionServer::tableCb(const TableArray&)’:
/home/imr/catkin_ws/src/turtlebot_arm/turtlebot_arm_object_manipulation/src/object_detection_action_server.cpp:315:16: error: ‘pose2str3D’ is not a member of ‘mtk’
                mtk::pose2str3D(table_pose).c_str(),
                ^
/home/imr/catkin_ws/src/turtlebot_arm/turtlebot_arm_object_manipulation/src/object_detection_action_server.cpp: In member function ‘int turtlebot_arm_object_manipulation::ObjectDetectionServer::addObjects(const std::vector<turtlebot_arm_object_manipulation::ObjectDetectionServer::DetectionBin>&)’:
/home/imr/catkin_ws/src/turtlebot_arm/turtlebot_arm_object_manipulation/src/object_detection_action_server.cpp:337:28: error: ‘pose2str3D’ is not a member of ‘mtk’
                    bin.id, mtk::pose2str3D(bin.getCentroid()).c_str(), bin.countObjects(), CALLS_TO_ORK_TABLETOP);
                            ^
/home/imr/catkin_ws/src/turtlebot_arm/turtlebot_arm_object_manipulation/src/object_detection_action_server.cpp:351:28: error: ‘pose2str3D’ is not a member of ‘mtk’
                    bin.id, mtk::pose2str3D(bin.getCentroid()).c_str(), bin.countObjects(), obj_name.c_str());
                            ^
/home/imr/catkin_ws/src/turtlebot_arm/turtlebot_arm_object_manipulation/src/object_detection_action_server.cpp:356:36: error: ‘point2str3D’ is not a member of ‘mtk’
                  obj_name.c_str(), mtk::point2str3D(out_pose.position).c_str());
                                    ^
/home/imr/catkin_ws/src/turtlebot_arm/turtlebot_arm_object_manipulation/src/object_detection_action_server.cpp:400:88: error: no matching function for call to ‘moveit::planning_interface::PlanningSceneInterface::addCollisionObjects(std::vector<moveit_msgs::CollisionObject_<std::allocator<void> > >&, moveit_msgs::PlanningScene_<std::allocator<void> >::_object_colors_type&)’
       planning_scene_interface_.addCollisionObjects(collision_objects, ps.object_colors);
                                                                                        ^
/home/imr/catkin_ws/src/turtlebot_arm/turtlebot_arm_object_manipulation/src/object_detection_action_server.cpp:400:88: note: candidate is:
In file included from /home/imr/catkin_ws/src/turtlebot_arm/turtlebot_arm_object_manipulation/src/object_detection_action_server.cpp:57:0:
/opt/ros/indigo/include/moveit/planning_scene_interface/planning_scene_interface.h:122:8: note: void moveit::planning_interface::PlanningSceneInterface::addCollisionObjects(const std::vector<moveit_msgs::CollisionObject_<std::allocator<void> > >&) const
   void addCollisionObjects(const std::vector<moveit_msgs::CollisionObject>& collision_objects) const;
        ^
/opt/ros/indigo/include/moveit/planning_scene_interface/planning_scene_interface.h:122:8: note:   candidate expects 1 argument, 2 provided
In file included from /opt/ros/indigo/include/ros/ros.h:40:0,
                 from /home/imr/catkin_ws/src/turtlebot_arm/turtlebot_arm_object_manipulation/src/object_detection_action_server.cpp:35:
/home/imr/catkin_ws/src/turtlebot_arm/turtlebot_arm_object_manipulation/src/object_detection_action_server.cpp: In member function ‘void turtlebot_arm_object_manipulation::ObjectDetectionServer::addTable(const std::vector<geometry_msgs::Pose_<std::allocator<void> > >&, const std::vector<turtlebot_arm_object_manipulation::ObjectDetectionServer::TableDescriptor>&)’:
/home/imr/catkin_ws/src/turtlebot_arm/turtlebot_arm_object_manipulation/src/object_detection_action_server.cpp:466:14: error: ‘point2str3D’ is not a member of ‘mtk’
              mtk::point2str3D(co.primitive_poses[0].position).c_str(), table_poses.size());
              ^
/home/imr/catkin_ws/src/turtlebot_arm/turtlebot_arm_object_manipulation/src/object_detection_action_server.cpp: In member function ‘void turtlebot_arm_object_manipulation::ObjectDetectionServer::DetectionBin::addObject(object_recognition_msgs::RecognizedObject)’:
/home/imr/catkin_ws/src/turtlebot_arm/turtlebot_arm_object_manipulation/src/object_detection_action_server.cpp:632:37: error: ‘yaw’ is not a member of ‘mtk’
         yaw_acc                  += mtk::yaw(obj.pose.pose.pose) * obj.confidence;
                                     ^
make[2]: *** [turtlebot_arm/turtlebot_arm_object_manipulation/CMakeFiles/object_interactive_manip_action_server.dir/src/interactive_manipulation_action_server.cpp.o] Error 1
make[1]: *** [turtlebot_arm/turtlebot_arm_object_manipulation/CMakeFiles/object_interactive_manip_action_server.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[... cut ...]
make[2]: *** [turtlebot_arm/turtlebot_arm_block_manipulation/CMakeFiles/pick_and_place_action_server.dir/src/pick_and_place_action_server.cpp.o] Error 1
make[1]: *** [turtlebot_arm/turtlebot_arm_block_manipulation/CMakeFiles/pick_and_place_action_server.dir/all] Error 2
Linking CXX executable /home/imr/catkin_ws/devel/lib/turtlebot_arm_object_manipulation/object_manipulation_test_actions
[100%] Built target object_manipulation_test_actions
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

```

Zabot, 2017-09-25:
Make sure you're using the indigo branch `git checkout indigo-devel`. I was able to reproduce your issue when I tried compiling `kinetic-devel` (which is the default branch) on an indigo install.

Abduoit, 2017-09-25:
I made sure that I selected indigo branch before cloning it, 
anyway I did `git checkout indigo-devel` then it switched to indigo branch correctly. It compiles now thanks

closed 2017-09-25 by Abduoit