turtlebot/turtlebot_arm #32: How to control the arm ??? with moveit

Abduoit, 2017-10-02:
I run `arbotix_terminal ` and I detect my five servos

Then, I run `roslaunch turtlebot_arm_bringup arm.launch` until I see `[INFO] [1506951666.556013]: ArbotiX connected`

But when I run `roslaunch turtlebot_arm_moveit_config turtlebot_arm_moveit.launch`  I get the following error, Please tell me how to control the arm ?? by rviz or moveit ???

```
abdulrahman@abdulrahman-ThinkPad-X230-Tablet:~$ roslaunch turtlebot_arm_moveit_config turtlebot_arm_moveit.launch 
... logging to /home/abdulrahman/.ros/log/aa1bb15c-a77d-11e7-a80b-e09d31097f54/roslaunch-abdulrahman-ThinkPad-X230-Tablet-3150.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://abdulrahman-ThinkPad-X230-Tablet:35791/

SUMMARY
========

PARAMETERS
 * /joint_state_publisher/use_gui: False
 * /move_group/allow_trajectory_execution: True
 * /move_group/arm/planner_configs: ['SBLkConfigDefau...
 * /move_group/controller_list: [{'joints': ['arm...
 * /move_group/gripper/planner_configs: ['SBLkConfigDefau...
 * /move_group/jiggle_fraction: 0.05
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_fake_contr...
 * /move_group/moveit_manage_controllers: True
 * /move_group/octomap_resolution: 0.025
 * /move_group/planner_configs/BFMTkConfigDefault/balanced: 0
 * /move_group/planner_configs/BFMTkConfigDefault/cache_cc: 1
 * /move_group/planner_configs/BFMTkConfigDefault/extended_fmt: 1
 * /move_group/planner_configs/BFMTkConfigDefault/heuristics: 1
 * /move_group/planner_configs/BFMTkConfigDefault/nearest_k: 1
 * /move_group/planner_configs/BFMTkConfigDefault/num_samples: 1000
 * /move_group/planner_configs/BFMTkConfigDefault/optimality: 1
 * /move_group/planner_configs/BFMTkConfigDefault/radius_multiplier: 1.0
 * /move_group/planner_configs/BFMTkConfigDefault/type: geometric::BFMT
 * /move_group/planner_configs/BKPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/BKPIECEkConfigDefault/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/BKPIECEkConfigDefault/type: geometric::BKPIECE
 * /move_group/planner_configs/BiESTkConfigDefault/range: 0.0
 * /move_group/planner_configs/BiESTkConfigDefault/type: geometric::BiEST
 * /move_group/planner_configs/BiTRRTkConfigDefault/cost_threshold: 1e300
 * /move_group/planner_configs/BiTRRTkConfigDefault/frountier_node_ratio: 0.1
 * /move_group/planner_configs/BiTRRTkConfigDefault/frountier_threshold: 0.0
 * /move_group/planner_configs/BiTRRTkConfigDefault/init_temperature: 100
 * /move_group/planner_configs/BiTRRTkConfigDefault/range: 0.0
 * /move_group/planner_configs/BiTRRTkConfigDefault/temp_change_factor: 0.1
 * /move_group/planner_configs/BiTRRTkConfigDefault/type: geometric::BiTRRT
 * /move_group/planner_configs/ESTkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/ESTkConfigDefault/range: 0.0
 * /move_group/planner_configs/ESTkConfigDefault/type: geometric::EST
 * /move_group/planner_configs/FMTkConfigDefault/cache_cc: 1
 * /move_group/planner_configs/FMTkConfigDefault/extended_fmt: 1
 * /move_group/planner_configs/FMTkConfigDefault/heuristics: 0
 * /move_group/planner_configs/FMTkConfigDefault/nearest_k: 1
 * /move_group/planner_configs/FMTkConfigDefault/num_samples: 1000
 * /move_group/planner_configs/FMTkConfigDefault/radius_multiplier: 1.1
 * /move_group/planner_configs/FMTkConfigDefault/type: geometric::FMT
 * /move_group/planner_configs/KPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/KPIECEkConfigDefault/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/KPIECEkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/KPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/KPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/KPIECEkConfigDefault/type: geometric::KPIECE
 * /move_group/planner_configs/LBKPIECEkConfigDefault/border_fraction: 0.9
[... 173 lines cut ...]
* MoveGroup using: 
*     - ApplyPlanningSceneService
*     - ClearOctomapService
*     - CartesianPathService
*     - ExecuteTrajectoryAction
*     - GetPlanningSceneService
*     - KinematicsService
*     - MoveAction
*     - PickPlaceAction
*     - MotionPlanService
*     - QueryPlannersService
*     - StateValidationService
********************************************************

[ INFO] [1506954416.216574010]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1506954416.216614559]: MoveGroup context initialization complete

You can start planning now!

terminate called after throwing an instance of 'Ogre::Exception'
  what():  OGRE EXCEPTION(5:): Could not find font Liberation Sans in MovableText::setFontName
[rviz_abdulrahman_ThinkPad_X230_Tablet_3150_8056263447110076151-4] process has died [pid 3182, exit code -6, cmd /opt/ros/kinetic/lib/rviz/rviz -d /home/abdulrahman/catkin_ws/src/turtlebot_arm/turtlebot_arm_moveit_config/launch/moveit.rviz __name:=rviz_abdulrahman_ThinkPad_X230_Tablet_3150_8056263447110076151 __log:=/home/abdulrahman/.ros/log/aa1bb15c-a77d-11e7-a80b-e09d31097f54/rviz_abdulrahman_ThinkPad_X230_Tablet_3150_8056263447110076151-4.log].
log file: /home/abdulrahman/.ros/log/aa1bb15c-a77d-11e7-a80b-e09d31097f54/rviz_abdulrahman_ThinkPad_X230_Tablet_3150_8056263447110076151-4*.log

```

Zabot, 2017-10-03:
This looks like an issue with rviz. Try removing the rviz launch from `turtlebot_arm_moveit.launch` and see if the launch still crashes. If it doesn't try reinstalling rviz.
```
  <!-- Run Rviz and load the default config to see the state of the move_group node -->
  <include file="$(find turtlebot_arm_moveit_config)/launch/moveit_rviz.launch">
    <arg name="config" value="true"/>
    <arg name="debug" value="$(arg debug)"/>
  </include>
```

Abduoit, 2017-10-03:
Thanks @Zabot 

Now, I got rviz and moveit. BUT when I change the virtual arm and press `plan and execut`e, the actual arm is not moving. ??

corot, 2017-10-06:
Well, many different things can be happening here...  can you provide more details, mostly the log. Basic checks:

- [ ] Do you see the arm in RViz with its real physical pose? (otherwhise said, are joints state properly published?)
- [ ] If not, are the motors really powered? Or is the arbotix driver properly connected with the board. Nice way to test this is the arbotix GUI, that allows you to control each motor individually
- [ ] Is the trajectory controller available and working? (you can directly call the action server using for example rosrun actionlib axclient.py
- [ ] Is MoveIt handling the trajectory controller? (check for example with rostopic info the action goal, if MoveIt is publishing to it

Abduoit, 2017-10-09:
@corot 

-The arm in rviz with its real physical position, I moved the physical arm and I saw its moving in rviz. 
- I could control individual joint by arbotix_gui
- for actionlib axclient.py, I run 
I got window 
-

corot, 2017-10-09:
I mean, you can check if moveit publishes on your arm contoler action server, e.g.
```
rostopic info /arm_controller/follow_joint_trajectory/goal
```
if so, echo the goal when you send one on RViz
```
rostopic echo /arm_controller/follow_joint_trajectory/goal
```
For sending a goal to the controller, you can use
```
rosrun actionlib axclient.py /arm_controller/follow_joint_trajectory
```
but it's true it will be very tedious to specify a joint trajectory, so better try to make RViz MoveIt plugin work, or try to command the arm using code. You can find an example in python [here](https://github.com/turtlebot/turtlebot_arm/blob/kinetic-devel/turtlebot_arm_moveit_demos/bin/pick_and_place.py).

huseyin26es, 2018-01-31:
Hi 
Did you solve this problem?  I have a same problem with yours, i coould not still handle it.

corot, 2018-01-31:
Hi, I cannot fix something I cannot reproduce. Please follow the checks in my previous comment and report here the outcome.

huseyin26es, 2018-01-31:
rostopic info /arm_controller/follow_joint_trajectory/goal

output : 
Type: control_msgs/FollowJointTrajectoryActionGoal

Publishers: None

Subscribers: 
 * /arbotix (http://huseyin:45015/)

corot, 2018-01-31:
So MoveIt! is not connected to the arm controller action server. Any error in the log?

huseyin26es, 2018-01-31:
now , i can control arm with arbotix_gui command. What exactly do you mean with log ?

huseyin26es, 2018-01-31:
and why moveit not is connecting arm, i think i did everything  correct.

corot, 2018-01-31:
Well,,,
  Publishers: None
means MoveIt! is not sending goals to the arm

arbotix_gui is different because it talks directly to every joint, not the FollowJointTrajectory controller

huseyin26es, 2018-01-31:
yes , i understand now. What should i do now for controlling in MoveIt! ?

corot, 2018-01-31:
Unless there's an error, it should just happen. Check the log for errors. I'll take a look later. You launch :

`roslaunch turtlebot_arm_moveit_config turtlebot_arm_moveit.launch 
`

right?

corot, 2018-01-31:
Hi, I think I found your problem; you must add sim:=false to your launch command. If not, you are using MoveIt! just to command a "virtual" arm, not your real one. So try to launch:

> roslaunch turtlebot_arm_moveit_config turtlebot_arm_moveit.launch sim:=false

huseyin26es, 2018-02-01:
Yes you are right, i did and it moves the robot :D Thanks a lot .

closed 2018-05-02 by corot