turtlebot/turtlebot_arm #35: Not possible to move (real) gripper of Pincher Arm

joellutz, 2018-04-23:
I've freshly cloned & built the turtlebot_arm & the [arbotix](https://github.com/vanadiumlabs/arbotix_ros) repository (I had to remove the object_manipulation folder to get my build done, but I don't use it anyway). My physical Pincher Arm is connected to /dev/ttyUSB0. In order to move it around with MoveIt & Rviz, I had to change in the `turtlebot_arm_moveit_config/demo.launch` file the `simulation` arg for the arm bringup launch file to `false` as shown here:
`<include file="$(find turtlebot_arm_bringup)/launch/arm.launch"> 
    <arg name="simulation" value="false"/>
  </include>`
When I now start the MoveIt demo with `roslaunch turtlebot_arm_moveit_config demo.launch ` I can plan & execute and the pincher arm moves to the goal. But if I change the Planning Group in Rviz to "gripper" instead of "arm" (under "Planning Request" in Rviz), I can still plan & execute, the gripper moves in Rviz and the logging says the Gripper Controller succeeded. But my real robot gripper doesn't move a bit, even with various different positions.

Every second I also get a log like this:
`[ WARN] [1524497161.302061425]: The complete state of the robot is not yet known.  Missing gripper_joint`
I suspect my error has something to do with this. Also in Rviz, only one of the two gripper parts can be moved, while on the real pincher robot arm, it is a parallel gripper where both parts move symmetrically.

Can someone help me with this? Thank you!
Cheers

joellutz, 2018-04-27:
As I mentioned, the gripper of the Pincher arm differs from the real robot. In addiditon, I had to modify the turtlebot_arm_moveit_config/config/pincher_arm.srdf file in order to get my pincher arm (and gripper) move around (in Gazebo):

- add a `<passive_joint name="gripper_link_joint" />`
- rename the base_link of the kinematic chain of the planning group `arm` from `base_link` to `arm_base_link` (as it is in the turtlebot_arm.srdf file), so the whole group tag looks like this:

`<group name="arm">
        <chain base_link="arm_base_link" tip_link="gripper_link" />
    </group>`

I don't know if this is necessary or not, but maybe someone finds this difference from the pincher_arm.srdf to the turtlebot_arm.srdf file interesting/useful.
Note that I didn't have to do this when moving the real pincher arm, but rather to move it around in Gazebo. I currently can do the latter only with a self-written control-package and not with the arbotix controllers, so maybe this modifications aren't necessary when using the arbotix controllers.

corot, 2018-04-29:
Hi!
For the first point, there's a [PR](https://github.com/vanadiumlabs/arbotix_ros/pull/28) opened ages ago to make arbotix work with the pitcher. So would be great if you can put it up-to-date and ping arbotix maintainers to merge it so it makes it way to the next release!
I'm re-reading my documentation and it says: _To use the PhantomX Pincher .... You will need arbotix_ros version 0.11.0_ What is funny cause last released version is 0.10.0. So I suppose this means that you need a for integrating the above mentioned PR.

For the malformed gripper, please try [this PR](https://github.com/turtlebot/turtlebot_arm/pull/24). By using it. the gripper appears in RViz in the right place. But I cannot tell if it works because I don't have a real Pitcher nor works on Gazebo (as you explained). I didn't merge the PR cause used to conflict somehow with kobuki plugin (read my comments), but it seems to work now with Gazebo9. So if you try it and works, I will merge.

joellutz, 2018-05-03:
Hi!
I'm already using the pull request version of [arbotix_ros 0.11.0](https://github.com/corb555/arbotix_ros) you mentioned, otherwise I couldn't move the real pincher arm.
Regarding the other [PR](https://github.com/turtlebot/turtlebot_arm/pull/24), I've adapted my files according to the changes suggested in the PR, including the `mechanicalReduction `set to 2 in the `transmission`-tag for the `gripper_joint`. I can see an improvement when planning & executing gripper positions (grip_open, grip_closed etc.) in Rviz: both fingers are now moved when selecting a goal state, even though the gri_open position seems too wide:
[image]
When looking at it in Gazebo, sometimes opening/closing the gripper works for one finger (the `gripper_active_link`), but the other one (`gripper_active2_link`) stands still like before integrating the PR:
[image]
On other occasions, the execution fails with the error message 

> Controller is taking too long to execute trajectory (the expected upper bound for the trajectory execution was 0.985854 seconds). Stopping trajectory.

Additionally, the `gripper_active_link `doesn't seem to be correctly attached, as it glitches away from the gripper ground plane, mostly after moving the arm:
[image]

I have to mention here that I currently don't use any arbotix controllers at all when simulating the arm in Gazebo (only when moving the real arm, which I don't have problems with anymore), because I couldn't get it to work yet. Instead, I'm using a tinkered control-package (can be found [here](https://github.com/joellutz/pincher_arm/tree/master/turtlebot_arm_control)) and I had to slightly modify the `turtlebot_arm_moveit_config/config/controllers.yaml` file (see [here](https://github.com/joellutz/pincher_arm/blob/master/turtlebot_arm_moveit_config/config/controllers.yaml)), so maybe some of my issues come from this rather ugly workaround.
This malformed gripper isn't that much of an issue for my project, as long as I can move the arm in Gazebo, but it would still be nice to have a "correct" gripper in gazebo.

corot, 2020-10-26:
Hey @joellutz, most probably this comes too late, but I just noticed that these people are using the [Pincher arm](https://github.com/Interbotix/phantomx_pincher_arm/tree/master/phantomx_pincher_arm_description/urdf) and have their own repo. As they are selling the arm,  I would expect that the description is correct,