turtlebot/turtlebot_arm #40: Phantom X Pincher Arm on ROS Melodic/Noetic

pytalha, 2020-11-03:
I want to run the good old Turtlebot PhantomX Pincher Arm on ROS Melodic/Noetic. I'm currently running ROS Melodic on Ubuntu 18.04 and have been facing a lot of dependency issues. I can't open arbotix_gui, can't execute my motion plan on Rviz and can't find a working Gazebo launch file. 

If anyone is still working with pincher arm on ROS Melodic/Noetic, please guide.

corot, 2020-11-04:
I'm using the default turtlebot arm (with a static finger) in melodic and all works fine. But I'm not using arbotix anymore, but [dynamixel_motor](http://wiki.ros.org/dynamixel_motor?distro=kinetic) and the  [USB2AX](https://www.generationrobots.com/en/401584-usb2ax-for-dynamixel-servos.html) to interface the motors.

pytalha, 2020-12-23:
Is there a repo for further guidance? Would appreciate if you could walk me through the steps you took to use the Turtlebot arm without arbotix.

pytalha, 2020-12-23:
> I'm using the default turtlebot arm (with a static finger) in melodic and all works fine. But I'm not using arbotix anymore, but [dynamixel_motor](http://wiki.ros.org/dynamixel_motor?distro=kinetic) [...]

Is there a repo for further guidance? Would appreciate if you could walk me through the steps you took to use the Turtlebot arm without arbotix.

bjsowa, 2021-03-26:
@pytalha I don't know if you resolved your issues or gave up, but if you are still looking for a solution, you can try these repository instead:
https://github.com/fictionlab/pincher_arm

The `arbotix_gui` is broken since Melodic if I remember correctly.

corot, 2021-03-29:
Nice job, @bjsowa. Yours is defintly the reference repo to use the pitcher.  I just link here my fork of tb_arm cause I did recently some improvements that may be useful to port for the pitcher:
https://github.com/turtlebot/turtlebot_arm/compare/melodic-devel...corot:melodic-devel
Mainly, I moved the gripper_link within the gripper paddles. I didn't bother to merge into tb_arm upstream cause I though that nobody was using it!  But i'll do gladly if anyone is interested (actually, I posted the question in #38)

bjsowa, 2021-03-30:
I actually made the URDF for `pincher_arm_description` mostly based on your fork, but I reworked the gripper description to fit the way I control it.

I assumed the `turtlebot_arm_description` is licensed under BSD (the `package.xml` says so), so I would appreciate if you could resolve #41 so I could reproduce the copyright notice in my packages.

pytalha, 2021-04-07:
> @pytalha I don't know if you resolved your issues or gave up, but if you are still looking for a solution, you can try these repository instead:
> https://github.com/fictionlab/pincher_arm
> 
> The `arbotix_gui` is broken since Melodic if I remember correctly.

Does that mean that it won't work with arbotix controller? Could you please specify the controller if it needs to be replaced?

bjsowa, 2021-04-16:
The `pincher_arm` packages use the Arbotix controller. It's just the `arbotix_gui` executable that is broken (see vanadiumlabs/arbotix_ros#49) so you won't be able to control the arm this way for now.