turtlebot/turtlebot_arm #41: Add a License file

bjsowa, 2021-01-08:
Could you please add a license file to the repository?

corot, 2021-03-31:
I'm rather ignorant about OS licensing, but I suppose I can just copy the one from turtlebot: https://github.com/turtlebot/turtlebot/blob/melodic/LICENSE
@mikeferguson , @mmwise , @stonier is this right?

mikeferguson, 2021-03-31:
I think you could basically adapt the headers from https://github.com/turtlebot/turtlebot_arm/blob/melodic-devel/turtlebot_arm_kinect_calibration/src/detect_calibration_pattern.cpp (or any of the other early files in the repo)

stonier, 2021-04-01:
Aye, approximately what @mikeferguson said. Ground truth of these things is always at opensource.org, e.g. [BSD-3](https://opensource.org/licenses/BSD-3-Clause). 

* Bundle a LICENSE file at the root (easy to discover)
* Copy it into each header file (opinions vary on if this is necessary, but it seems prudent to have some hint for people in the publicly distributed files)
* Make sure you set your copyright fields appropriately

corot, 2021-04-04:
done

closed 2021-04-04 by corot